RolfZuckowskiUltras/TwoTrees-Sapphire-Pro-Plus-Marlin2.0-with-GUI #1: Malfunction on Sapphire Pro

piberry, 2021-01-31:
Hello!

After trying a lot of different firmwares i was finally glad to find this promising one.
I have some Problems. I´m using 4x TMC 22009 in standalone mode on the printer. 

-The move section does nothing at all, all axes.
-When homing, x and y move but the z stepper just makes some high pitch noise and doesn´t move.

In Marlin i set Sapphire Pro and of course the 2209 drivers, standalone. I made the correct settings on both sections,  custom and the other one so it must definitely work.
Using the precompiled bin, i have the same problems. It´seems whatever setting i make for the stepper drivers just gets ignored.
it´s a pity because the firmware looks very interessting beside this.

Can you have a look please? Let me know if i can help in any way.

RolfZuckowskiUltras, 2021-02-16:
I ordered some TMC2209, they arrived today. I'll test that issue.

RolfZuckowskiUltras, 2021-02-17:
TMC2209-Steppers seem to work without a problem in standalone mode for me.

There are two things to look out on the hardware-side. Compared to the X+Y-Drivers (TMC2208), the Z-Axis is driven (factory setup) by a A4988. TMC220# and A4988 turn in different directions. So you probably have to invert the Z-Axis.

If your Z-Axis has not moved at all (assumption you did not start it at the lowest point) I would additionally make shure, that the vref voltage is correct.

For now the lvgl-gui seems to have the issue of not loading the correct icons (see other Issue). Is that existant for you too?

closed 2021-02-17 by RolfZuckowskiUltras

piberry, 2021-03-20:
Hi!
I tried again today from scratch. All problems i pointed out above are still present in this firmware. Inverting the z-axis does nothing but the same noise but the axis wont move. Also the manual move section is completely dead.
Very odd...

RolfZuckowskiUltras, 2021-03-20:
Hey,
manual movement is enabled after homing only. This configuration is made, since there is a potential of crashes with the Z-Spindle mounts, if a E3D Hemera is used. This is a one fits all configuration, since it won't hurt for the other machines. Manual movement works for me after homing.

I switched to two TMC2209_Standalone drivers on my Sapphire Plus and it works fine. Configurations were done in the custom section. This is all I can say from my personal point of knowledge 

If you really want help from the community or me, you have to post some further information or do some tests of the Z-Axis.
- Post your custom driver configuration section
- Post a video of the behaviour
- Interchange the driver Z<-->XY to see, if one is broken.
- Connect by USB-Serial to your Printer to directly control the Z-Stepper for diagnosis
--> If you switch on your drivers and turn it by hand: Is it as hard to turn as the XY-Ones? --> Wrong Vref or Driver enable State
--> Feel of the Stepper if the driver trys to turn

piberry, 2021-03-20:
I see...
When i press the z-endstop with a finger while homing, the noise stops and the move section is enabled. i can then move the z-axis totally normal  by menu. I tried all the differnt steppings, they all work fine. When i autohome or home-z, it goes crazy. The Printer will also hold after the motor done it´s thing.
I set the vref for all drivers to 0.9 V. I have tried to set it a little higher for the z-axis but this doesn´t do the trick. 
Since i didn´t know what else to do, i changed to UART mode (involves soldering) to see if it makes a difference. Well it doesn´t...
Next thing i tried was attaching a brand new motor on that cable. Same weird noise when homing. I also swapped drivers but the problem persists.
When the motor normally would spin up, there is a kind of sinus tone. the shaft is not moving but it can be moved a bit while acting. When the is motor on but not acting, it can not be moved easily.
I´ll do a video when the light is better tomorrow.

RolfZuckowskiUltras, 2021-03-20:
Ok. So movement does work, but not Homing. Look for your homing feedrate and or the maximal feedrate of your Z-Axis. This is probably too high. Following that one, the motor is not able to keep up with the speed.

piberry, 2021-03-21:
I changed the z homing feedrate to 10*60. It works now!
Thank you!
```
// Homing speeds (mm/min) (X,Y,Z)
#define HOMING_FEEDRATE_MM_M { (50*60),(50*60),(10*60)}  //custom, default: { (50*60),(50*60),(40*60)}
```