RolfZuckowskiUltras/TwoTrees-Sapphire-Pro-Plus-Marlin2.0-with-GUI #2: "No data" issue on Sapphire Plus

Beeblebrox17, 2021-02-09:
Hello!
I have tried compiling the binaries myself. When that didn't work, I tried the prcompiled binaries from the "release" section. I always encounter the following issue:
The booting starts, and then the display calibration shows up. Then buttons appear, but they all simply carry the lable "No data".
Could you give me any assistance as to what might be the cause?
Thank you!
Micha

RolfZuckowskiUltras, 2021-02-16:
I did a complete reload of the firmware with the gui an can confirm it. It did not occur to me, since I did not update the gui before. Seems to be a firmware (not GuI-Related) issue. I'll take a look.

RolfZuckowskiUltras, 2021-02-16:
The issue seems to be a little weird and very likely related to the original MKS repo or my compilation. The issue is existant out of the box with the mks repo. I'm honestly a little confused about that.

RolfZuckowskiUltras, 2021-02-16:
I have not cloue about that problem. I'll probably make up a issue at the original MKS-Firmware.

RolfZuckowskiUltras, 2021-02-17:
Confirmed.

Solution: The GuI is loaded over the assets folder. When compiling yourself via PIO, you are getting your **output in the folder .pio/mks_robin_nano35**

There is a folder called **/assets/**. That one contains the GuI. The firmware is located in the file **Robin_nano35.bin**. Just pull them onto your SD-Card and Reboot the printer. Precompiled Binaries will be fixed.

RolfZuckowskiUltras, 2021-02-17:
BTW: They must have changed it or this was just me beeing blind :-)

closed 2021-02-17 by RolfZuckowskiUltras