RolfZuckowskiUltras/TwoTrees-Sapphire-Pro-Plus-Marlin2.0-with-GUI #3: YMAX Problem

ElDiablo830, 2021-03-21:
Hello
I'm sorry i dont know if it is the right way to ask for help..

I have a SAPPHIRE PLUS (belt Z), my stock board has been replaced by MKS NANO V3 (due to death of v1.2)
MKS ROBIN NANO V3 kit with TS35 v2 screen, 5 x TMC 2209

I tried this firmware (thanks for that btw) realease's bin, i got some problems, i managed to fix myself on platformio.
But there is one that i can't solve..

I have stock endstop switch on XMIN/YMIN/ZMIN, i defined the printer as SAPPHIRE PLUS (with stock firmware settings as 300x300 bed) but it seems the printer is looking for Y max endstop. it doesnt stop and go away crush above 300..
I found that //#define USE_XMAX_PLUG
So, as i dont have MAX enstops i tried to undefined theses :
//#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG
but cant compile it seems that others stuff need it.
I dont know if it is the setting that can help anyway.

I tried other firmware and i have the same problem by the way.
before adding 3 max endstop, i try to ask you some advice

I don't know what the point, if bed size is set, it should can't overpass the bed limit..
I tried to reduce bed size and it still go crush out..

Thank for your work and help

RolfZuckowskiUltras, 2021-03-21:
Hi,
how can you face firmware issues, if you are not able to compile? The procompiled .bins are for V1.X-Boards only.

### Compilation Problems
There are changes made to the board, so you have to also do those in you compilation environment and configuration.h-File.

- Take a look into the description of my repository. The bottom part is copied from MKS and it describes the changes, that have to be normally done to the MKS Repo to use it. 

- [ ] Look out for the changes, needed for the Robin Nano V3.X
- [ ] Reverse the changes, that have to be done for the Robin Nano V1.X, but not for the V3.X
- [ ] Try to compile. When facing some .json-Errors, try to compile a second time.

### Min/Max Stuff
#### Enstop Stuff
It's a little hard to understand your problem. The wrong allocation of the endstops can be caused by changes to the pin-mapping on the board (V1.2 vs V3.0). The board-config in your compilation environment normally takes care of that. So no word from me on that, before your able to compile.

A aditional word on that: On some Sapphire-Printers, the X-Enstop logic is inverted. If you are facing a homing failure (when X-axis homes), just intervert that.

- [ ] I would recommend you to start with the configuration from scratch to avoid errors, that are generated by try and error approaches before.
- [ ] Consider doing a fork of my repo to show up the fixes to other people and easily merge in updates.
- [ ] When compiled from scratch, connect to the printer via USB-Serial-Terminal (Putty...) and find out the connection state of the endstops with the gCode [M119](https://marlinfw.org/docs/gcode/M119.html)

#### Homing Direction
Think about looking onto the front display. Your Point X0; Y0 is on the front left. So you are homing on X_Min and Y_Max. Take a look into the board manual, if your enstopps are connected into the right sockets.

The Bed-Size is used for the Soft-Endstops. Shortly saird: It does not go past the bedsize by software. The printer has to be homed, before it "knows" it's position.

TMC2209 --> Please remember. Those drive direction to be inverted, comparted to the A#### drivers.

### Further Support
Try to compile first and rethink your enstop-stuff after that from zero. It's hard to do guesses on compilations problems without the project or the log/errors at least. So please try to post more details next time.

ElDiablo830, 2021-03-22:
Hello thanks for the answer.
I'm native french speaker so my english is not so good.

I tried to explain that i've tried the precompiled .bin first, then i understood it was for v1.X board.
Then i install platformio and all the necessary, imported the marlin folder, i made adjustments recommended  for V3 board as explained on github details, defined as SAPPHIRE PLUS and succeed to compile many times to make some tests.

On V1.2 nano board, the home point was (when i face the printer's screen) at the opposite left border (back side of printer).
On this firmware, native homing go at the same point too.
But when i use the manual leveling fuction (for example, i guess the travel by starting a print will also exceed the bed), the first point come from home to me (front side of printer) but doesnt stop.

i guess this problem is also on X and Z.
It seems to be that the Y/X/Z MAX limit is not set.
My mechanical endstop switch are like stock one, on X/Y/Z MIN.
The MIN limits works well, but it seems travel doesnt stop like there is no X/Y/Z MAX limit.

That is a try to better explain the problem.
I'm looking for maybe a settings to define the X/y/z max travel limit ? To stop for not exceed the bed surface.

I will look asap all you suggest me, to solve or add some precision, thank you for the help.
I replace the board by connecting everything at same as V1.2 everything matched.
I will check with MKS data if everything is at the right place, and all you said.
I will let you know is i solve it
Thanks

Printer details :
SAPPHIRE PLUS V1
Dual Z belt sync
Stock stepper motors
Stock mechanical endstop switchs
Stock frame, linear and bracket
Dragon Hotend
Stock BMG extruder (no direct drive yet)
Heat bed 220v regulated

**MKS V3 ROBIN BUNDLE, included :**
MKS V3 ROBIN NANO
5 units TMC2209
MKS WIFI MODULE
TS35 V2 screen

RolfZuckowskiUltras, 2021-03-28:
Hey,
have you already solved your problem? Soft endstops are normally activated, if not manually diabled. The travel limits are automatically set by your bed size.

As far, as I understand, you seem to have the wrong Idea about the homing direction. Lets review my first comment.

> Homing Direction
>
>Think about looking onto the front display. Your Point X0; Y0 is on the front left. So you are homing on X_Min and Y_Max. Take a look into the board manual, if your enstopps are connected into the right sockets.

So shortly said, your X-Endstop is a X_Min one, since you are homing at X=0.0mm. Your Y-Endstop is a Y_Max one, since you are homing at Y≈300mm. Z gets Homed at Z=0.0mm so your Z-Endstop is a Z_Min one.

closed 2021-08-20 by RolfZuckowskiUltras