RolfZuckowskiUltras/TwoTrees-Sapphire-Pro-Plus-Marlin2.0-with-GUI #4: black screen

funaro13, 2021-04-04:
Congratulations for your work, I will use sapphire pro, it was compiled without any error, but then start the screen is black, the assets folder is not being installed, I tried to initialize by placing only the assets folder but it does not do the erase pictures

kenniesdead, 2021-07-09:
not sure if its helpful at all, but i got a black screen after install on my pro after enabling motion new so had to stay with classic jerk

RolfZuckowskiUltras, 2021-08-20:
Hey,
sorry for the late Answer. Did not get any messages. Have you put the firmware AND assets-folder onto the SD-Card?

`
  -  Enter the .pio\build\mks_robin_nano35 directory, copy the assets folder and Robin_nano35.bin to the sd card
  -  Insert SD card is to the motherboard, and you can see the update interface after power on.
`

If thats the case, you should try to set back the differences you did in comparison to the default step by step. I did also experience some weird issues, after enabling BL-Touch with GUI, which were fixed by SD-Read-Only Mode. So you may wanna try those steps and the SD-Thing as your last resort.

closed 2021-09-28 by RolfZuckowskiUltras