RolfZuckowskiUltras/TwoTrees-Sapphire-Pro-Plus-Marlin2.0-with-GUI #17: LCD Not displaying correctly for Sapphire Pro

princeje, 2021-11-22:
Followed all the steps to build this for a Sapphire Pro and the LCD is not displaying correctly. It looks like this:

[image]
k5
[image]

Configuration.h: https://pastebin.com/Xzj3XrLH

Sapphire Pro Specs:
- MKS Robin Nano v1.2
- MKS TFT35 V1.0
- BL Touch
- TMC2208 For X, Y, and E0
- A4988 for Z

RolfZuckowskiUltras, 2021-12-30:
Hey,
a another weird MKS-LVGL GUI Issue, since you got the same hardware as myself :-). Please try to compile the firmware with the Marlin Touch or Color GUI as a quick-check. If it does work, it's an issue of the LVGL-Gui. If it's the same, your screen seems to have different parameters. You can fix it by finetuning the offsets in the advanced configuration of marlin. It would be kind of you to share your changed parameters for other users in that case.

James-Ackerman, 2022-01-14:
I had a similar issue with the white bar being vertical and messing up the rest of the images. It turns out the issue was with the X_OFFSET (40.5mm) I had set for the BlTouch. I set it back to zero and the screen worked again. I then just set the offset through the GUI after re-flashing.

RolfZuckowskiUltras, 2022-01-14:
@James-Ackerman Thx. This is a hot hint. It's very likely, that marlin is using the same precompiler definitions for the display and BL-Touch. The Display is in the advanced config.h and the BL-Touch in the normal one. So the compiler does use the defintions globally and for me not. This can be cause through a version difference or different compilers. You're using PlatformIO with MinGW too, I guess?

closed 2022-02-19 by RolfZuckowskiUltras