RolfZuckowskiUltras/TwoTrees-Sapphire-Pro-Plus-Marlin2.0-with-GUI #18: New TwoTrees Sapphire models

herbstleyd, 2021-12-20:
There is a new version of the TwoTrees Sapphire Plus with dual-Z motors an Belt-Sync with only one Z-Stop
There is a Robin Nano 1.2 installed with all 5 TMC2225 Steppers. X, Y, Z1 and Z2 and E 
In this Version the display is rotated 180°
Is thera a possibility to get this Firmware running on the new versions?
I can support you with information about this new version if needed.
I ordered a Hemera for my Sapphire Plus and looking for a preconfigured Firmware.
Thank you very much!

RolfZuckowskiUltras, 2021-12-30:
Moin,
now I start to get all those weird problems. TwoTrees seems to do more revisions, than my granny baking cakes or they just try to clear out their old stock. At least I couldn't image why they would switch back to the belt-synced concept, that did prove itself to be a lot worse than the standalone-dual drive.

I saw, that you did start working on a own repo. I would recommend to maybe take a look at my playground repo, which is directly forked from marlin. I was quite not satisfied with the code-base of MKS since the LVGL-GUI seems to cause a lot of issues and sometimes does freeze up the printer while loading different menus.

I plan to move all of the automatic configurations to a direct forked marlin repo and to create different branches for the different board versions. A second repo is planned for providing a firmware with direct web interface for the official MKS-Wifi Module (ESP3D). Unfortuatly I'm a little short on spare time, why this won't be happening soon.

herbstleyd, 2022-01-05:
Ei Gude,
thank you very much for the reply. Yes, TwoTrees seems a little bit unsecure about their printers. But, okay xD
Yes, i got managed to compile my own Firmware and started my own repo. No need to rush and no stress! My printer is printing very well to this time. 
I am new to this Mariln stuff. But it seems pretty managable.

I like the idea with the auto-config tool. I will think about it and i will contact you if i have something usefull to work with ;)

RolfZuckowskiUltras, 2022-02-17:
Hey, did make a new Repo with different Branches for different machines. So the maintainence should be easier now. Feel free to do a fork and may bring in your config via a pull-request.

closed 2022-02-17 by RolfZuckowskiUltras