RolfZuckowskiUltras/TwoTrees-Sapphire-Pro-Plus-Marlin2.0-with-GUI #19: Homing Speeds

mjbmz, 2021-12-23:
Hi,

for me the default Homing Speeds where much to high. While X and Y homed fast but ok, the Z motor screamed and moved no mm. I Have change these Values:

From:
#define HOMING_FEEDRATE_MM_M { (50*60),(50*60),(40*60)}

To:
#define HOMING_FEEDRATE_MM_M { (20*60),(20*60),(4*60)}

Thanks for your work

RolfZuckowskiUltras, 2021-12-30:
Hey,
thx for the issue. I'm going to lower the homing speed, the next time I do touch the code. How fast your Z-Axis can actually go, depends a lot on lubrication, since the spindles are quite loose and sometimes tend vibrate, casusing those to block. A lower feedrate works more reliable, so thats a better decision.

RolfZuckowskiUltras, 2022-02-17:
Fixed in my new Repo.

closed 2022-02-17 by RolfZuckowskiUltras