RolfZuckowskiUltras/TwoTrees-Sapphire-Pro-Plus-Marlin2.0-with-GUI #21: Issue with Preconfig for v1.3 board

Semparo-Taro, 2022-01-08:
Unsure if I'm doing something wrong, but I have the S-Plus with Dual Z with Belt Sync.  The board is v1.3.  When I try the precompiled file it starts the update but hangs on 100%
[image]
[image]

RolfZuckowskiUltras, 2022-02-09:
This might be cause by the board beeing a newer release than the actual precompiled files. So I maybe have to update the project with the files from marlin at some point at time, to make this work.

RolfZuckowskiUltras, 2022-02-17:
@Semparo-Taro You may wanna try out my new Repo. It does contain the newest stable Marlin revision.

closed 2022-02-19 by RolfZuckowskiUltras