RolfZuckowskiUltras/TwoTrees-Sapphire-Pro-Plus-Marlin2.0-with-GUI #22: Error activating BLTouch

donvito4ever, 2022-02-02:
When I compile the last version with BLTouch activate, throw an error and I can't compile it...

Edit the Configuration.h and activate #define BL_TOUCH
Compile and throw the error...

[image]
[image]
[Configuration&adv.zip](https://github.com/RolfZuckowskiUltras/TwoTrees-Sapphire-Pro-Plus-Marlin2.0-with-GUI/files/7989085/Configuration.adv.zip)

RolfZuckowskiUltras, 2022-02-09:
Hey,
when I did fit a BL-Touch onto my machine I did experience a lot of Issues, caused by the MKS-Firmware and some Pasted&Copied configurations of L3tspeak. Because of that I did eventually start up from ground to make a better working firmware. I have a project eventually on my list to make a clean repo for multiple machines out of my playground, but it's in the back of the line, since there is not that mutch plenty time to me.

Missing in your Project is the definition of the Pin, where the signal of your BL-Touch is mounted on. This depends on the plug you're using. If you do refer the manual of the MKS Robin nano it should be Pin PE6. 

-->`#define Z_MIN_PROBE_PIN PE6 // Pin 32 is the RAMPS default`

The [mentioned playground repo of mine](https://github.com/RolfZuckowskiUltras/Marlin_2.0.X_Bugfix_TwoTrees_Sapphire_Plus_DUAL_Z_HEMERA), is little bit dirty, but pretty mutch working fine and finetuned for a Sapphire Plus Dual-Z with Hemera. As I saw, you're using a Sapphire Plus with Hemera. So it might be the fastest way (for now) for you to just take out the Dual-Z-Stuff of this project and used that.

Just look for the lines with the configurations for the `SAPPHIRE_PLUS_DUAL_Z` in this project (config.h/config_adv.h) and change those accordingly.

Cheers!

RolfZuckowskiUltras, 2022-02-17:
@donvito4ever I created a new Repo with multiple branches for different Sapphire Plus-Versions. There are only the classic TMC2208/A4988-Versions in there yet, but you may wanna try it out. The Issue should also be solved in the new Repo, since there is not MKS-Stuff in it :-).

closed 2022-02-17 by RolfZuckowskiUltras